3DCoded/3MS #2: Using existing mainboard to control 3MS

florek41, 2024-07-16:
It would be useful to document the possibility of using the current/existing mainboard, as it only requires free space for stepsticks, making it much cheaper to set up.

Example:
I have a BTT Octopus on my printer and I'm currently using only 4 of the 8 available steppers, so there is no need to purchase a separate controller for 3MS.

3DCoded, 2024-07-16:
Since each mainboard has a different Klipper configuration, I've renamed this one to be specific to the BTT Octopus. I just updated the documentation and created a configuration for it. Something to note (this is in the documentation) is that you need a TMC2209 for each of the motor ports.

3DCoded, 2024-08-13:
@florek41 Let me know if there's anything I should change in the config for your setup.

3DCoded, 2024-09-11:
Closing this issue as its been open for 30 days with no activity. If this isn't resolved, feel free to reopen this.

closed 2024-09-11 by 3DCoded