3DCoded/3MS #4: ZoneStar Z9V5 Pro with ZM3E4 controller

kaiatkins, 2024-08-16:
3D Printer ZoneStar Z9V5 Pro

Controller: ZM3E4

https://github.com/ZONESTAR3D/Control-Board/tree/main/32bit/ZM3E4

Configs:

https://github.com/Z9V5PRO/Z9V5Pro-MK4-Klipper

<Printer.cfg>

[printer]
kinematics: corexy
max_velocity: 300
max_accel: 3000
max_z_velocity: 25
max_z_accel: 30

[stepper_x]
step_pin: PE1 
dir_pin: !PE2
enable_pin: !PE0
microsteps: 16
rotation_distance: 20
endstop_pin: !PC13
position_endstop: 0
position_max: 300
homing_speed: 50
homing_retract_dist: 10
second_homing_speed: 10.0

[stepper_y]
step_pin: PE5
dir_pin: !PE4
enable_pin: !PE6
microsteps: 16
rotation_distance: 20
endstop_pin: !PE3
position_endstop: 0
position_max: 300
homing_speed: 50
homing_retract_dist: 10
second_homing_speed: 10.0

[stepper_z]
step_pin: PD3
dir_pin: PD4
enable_pin: !PD2
microsteps: 16
rotation_distance: 4
position_endstop = -1.220
endstop_pin: !PD7
position_min: -20
position_max: 400
homing_retract_dist: 0
second_homing_speed: 8.0

[stepper_z1]
step_pin: PC12
dir_pin: PD0
enable_pin: !PC11
endstop_pin: !PD1
microsteps: 16
rotation_distance: 4

[screws_tilt_adjust]
screw1: 2, 16.5
screw1_name: front left screw
screw2: 293,16.5
screw2_name: front right screw
screw3: 293,286.5
screw3_name: rear right screw
screw4: 2,286.5
screw4_name: rear left screw
horizontal_move_z: 10
speed: 50
screw_thread: CW-M4

[extruder]
step_pin: PD10
dir_pin: !PD9 
# change PD9 -> !PD9
enable_pin: !PD11
microsteps: 16
rotation_distance: 7.71
gear_ratio: 50:17
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC5
sensor_type: Generic 3950
# NTC 100K B3960 (used)
sensor_pin: PC2
min_temp: 0
max_temp: 250
max_extrude_only_distance: 500
max_extrude_cross_section: 5
control = pid
pid_kp = 18.853
pid_ki = 0.539
pid_kd = 164.732

[verify_heater extruder]
check_gain_time:999999

[heater_fan hotend_fan]
pin: PB8
max_power: 1.0
fan_speed: 1
kick_start_time: 0
heater: extruder
heater_temp: 60.0

[heater_bed]
heater_pin: PA2
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC3
control = pid
pid_kp = 74.758
pid_ki = 1.181
pid_kd = 1183.043
min_temp: 0
max_temp: 80

[fan]
pin: PB1
max_power: 1.0

<multi_extruder.cfg>

[extruder_stepper extruder_1]
extruder: 
step_pin = PD13
dir_pin = !PD12
enable_pin = !PD14
microsteps: 16
rotation_distance: 7.71
gear_ratio: 50:17

[extruder_stepper extruder_2]
extruder: 
step_pin = PC6
dir_pin = !PD15
enable_pin = !PC7
microsteps: 16
rotation_distance: 7.71
gear_ratio: 50:17

[extruder_stepper extruder_3]
extruder: 
step_pin = PA8
dir_pin = !PC9
enable_pin = !PC10
microsteps: 16
rotation_distance: 7.71
gear_ratio: 50:17

3DCoded, 2024-08-16:
I've added support for the ZM384 [here](https://3dcoded.github.io/3MS/setup/controllers/zm384main/). Due to some quirks with Klipper, as of now, only three extruders will work with the 3MS, and the fourth one will be used as the main extruder.

kaiatkins, 2024-08-16:
Wow that was fast :) Thank you. I have ordered some parts for my machine and will be reworking it.

I plan on using the bambulabs splitter and attach to the hotend.

[image]

is the ok?

3DCoded, 2024-08-16:
That splitter looks good, but you can also print [this one](https://www.printables.com/model/541678-bambu-ams-no-catch-y-splitter-pc4-m10).

3DCoded, 2024-08-21:
The spitter I previously linked is no longer recommended for setups with more than two filament units (you would need three of those splitters). For a system like yours, [this one](https://www.printables.com/model/122982-mini-splitter-v2-for-mmu21s-mod-pc4-01-fitting-ver) is now recommended (you only need one of these splitters).

kaiatkins, 2024-08-24:
The ZoneStar Z9v5 printer has 4 BMG extruders on the back and feeds either into the company's 4 in 1 out (mixing hotend) or 4 in 1 out (non- mixing hotend) but they lack in quality. So I have switched to klipper and wanted the option to change the hotend into a Bowden single E3D v6 then use like the bambulabs AMS.
[image]
[image]

3DCoded, 2024-08-24:
The 3MS will work fine on an E3D v6 hotend. I recommend getting the splitter as close to the hotend as possible for the speed of tool changes.

kaiatkins, 2024-08-25:
What would be the optimal location for the filament sensor? Currently its the red on the back of the machine.

3DCoded, 2024-08-25:
The optimal location for the filament sensor is between the splitter and the hotend.

kaiatkins, 2024-08-26:
I picked up on sale this (SFS 2.0) that I will use. Once everything arrives, I will start :) thank you for your quick responses, guidance and support 👍
[image]

3DCoded, 2024-09-20:
@kaiatkins, is the configuration working for you? If not, can you explain what's happening so I can fix it?

kaiatkins, 2024-09-20:
I was just coming on to let you know, I am still waiting for 1 part I ordered to put together then will be ready to go. Thanks for checking in, I am excited to convert to 3MS :)

kaiatkins, 2024-10-07:
package for hotend lost in transit so seller is resending :(

kaiatkins, 2024-10-20:
hello, it has finally arrived and I have started to put the stuff together. I will make mount for the sensor and 4 to 1. I'm contemplating moving the BMG to the top back of the machine instead of the default install, so I can cut out some of the Bowden tube.

[image]

warm regards from Cambodia 🇰🇭

kaiatkins, 2024-10-22:
[image]
 Slowly getting there :)

3DCoded, 2024-10-22:
Thank you for keeping track of your build here. I'm sure it will help other people using a similar setup.

kaiatkins, 2024-10-27:
[image]
[image]

So it looks like I’m working with mainboard ZM3E4 v2.1 with TMC2225’s

Are the steppers ok to use or should I replace with 2209s? I don’t see UART so looks like stand-alone.

I am flashing klipper and installing the configuration as it comes with marlin natively

here is the flash information 

[image]

kaiatkins, 2024-10-27:
For my Klipper I use biqu pi which also use it for 2 machines
printer_3MS_data
printer_GEMINI_data

the klipperscreen is controlled in

/home/biqu/.config/KlipperScreen/KlipperScreen.conf

so I need to make sure I save the info to switch between the 2 printers

kaiatkins, 2024-10-27:
[image]

Dynamic Macros is giving me issue about path

I tried 
~/printer_3MS_data/config 
and full path
 /home/biqu/printer_3MS_data/config

kaiatkins, 2024-10-27:
Error loading template 'CHECK_FSENSOR': jinja2.exceptions.TemplateAssertionError: no filter named 'str'

I changed 
variable_fsensor_name: "**switch_sensor**" # name of filament_switch_sensor right before printer's extruder
[image]

kaiatkins, 2024-10-27:
I am also getting error 

Option 'config_path' is not valid in section 'dynamicmacros'

[dynamicmacros]
config_path: ~/printer_3MS_data/config # If you left your printer configuration path as the default, you don't need to specify this
configs: dynamic.cfg

[image]

3DCoded, 2024-10-27:
> Option 'config_path' is not valid in section 'dynamicmacros'

You no longer need to specify this option. DynamicMacros automatically finds the correct configuration path for your Klipper instance.

kaiatkins, 2024-10-27:
I was just following https://3dcoded.github.io/DynamicMacros/setup/ and copied it from there, but looks fine now since I removed

when I do sudo service klipper restart

[image]

3DCoded, 2024-10-27:
Thank you for catching that. I've updated the documentation to remove that option. 

> when I do sudo service klipper restart

Just restart your Klipper however you usually do, just not the `RESTART` command. You can do this from the power menu in Mainsail.
[image]

kaiatkins, 2024-10-27:
Testing the Dynamic Macros

If you run the command MYTEST, the output should be:  Hi! (Works)
Then I go back in file and change to Hi, Bong!, save no restart  Hi! (No Update)
but if I run DYNAMIC_MACRO then MYTEST Hi, Bong! (Works)

3DCoded, 2024-10-27:
Yes. That's expected behavior. `DYNAMIC_MACRO` reloads the macros without restarting Klipper.

kaiatkins, 2024-10-27:
[image]

What I was showing by following that, 

Save the file, but do not restart Klipper.

Run MYTEST again, it didnt update until I manually ran DYNAMIC_MACRO

kaiatkins, 2024-10-27:
> Error loading template 'CHECK_FSENSOR': jinja2.exceptions.TemplateAssertionError: no filter named 'str'
> 
> I changed variable_fsensor_name: "**switch_sensor**" # name of filament_switch_sensor right before printer's extruder
> 
> [filament_switch_sensor switch_sensor] switch_pin: PC8 pause_on_runout: False runout_gcode: PAUSE # [pause_resume] is required in printer.cfg M117 Filament switch runout insert_gcode: M117 Filament switch inserted
> 
> [filament_motion_sensor encoder_sensor] switch_pin: PB13 detection_length: 2.88 extruder: extruder pause_on_runout: False runout_gcode: PAUSE # [pause_resume] is required in printer.cfg M117 Filament encoder runout insert_gcode: M117 Filament encoder inserted
> 
> I have commented out 3MS to check if the SFS filament sensor was setup properly and seems to be working
> 
> <img alt="Screenshot 2024-10-27 at 4 16 43 PM" width="557" src="https://private-user-images.githubusercontent.com/2082800/380483313-40be7143-95e7-486b-9082-854f1465c318.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAwMzI0MzQsIm5iZiI6MTczMDAzMjEzNCwicGF0aCI6Ii8yMDgyODAwLzM4MDQ4MzMxMy00MGJlNzE0My05NWU3LTQ4NmItOTA4Mi04NTRmMTQ2NWMzMTgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjdUMTIyODU0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGRjNWM3NGY0ZTczMTUxMzI1NzM1YTU1OTgzMzM4ZGNiYjU2MzA0ZmE4ZDFlMTc5YWU2MTJjMGM1ZjZlNzI3OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.iGDjiDUeRySEbL9kcAJOijGGahS475pcuHkdQJeXglY"> <img alt="Screenshot 2024-10-27 at 4 17 01 PM" width="565" src="https://private-user-images.githubusercontent.com/2082800/380483331-09556bf4-e8f8-461c-a99c-349adeeffdf9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAwMzI0MzQsIm5iZiI6MTczMDAzMjEzNCwicGF0aCI6Ii8yMDgyODAwLzM4MDQ4MzMzMS0wOTU1NmJmNC1lOGY4LTQ2MWMtYTk5Yy0zNDlhZGVlZmZkZjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjdUMTIyODU0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjMzOWNmNmMxOGRjNWI4MWZhNTk1MjZkNjg1ZGE4MDhjMDU4ZWEzMTFmMWNiNjc0ODg1Yjg2YTIzNWYyZTYzOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.fBAnlPxk3QbFKii1aNfcwm1dtJseiMv8AA0xWCE6V8c">

here is the log after I reenabled 3MS
[klippy.log](https://github.com/user-attachments/files/17533903/klippy.log)

3DCoded, 2024-10-27:
> <img alt="Screenshot 2024-10-27 at 7 17 55 PM" width="713" src="https://private-user-images.githubusercontent.com/2082800/380495886-0e01ae5a-20e3-4146-a5d5-65ef879319b4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAwNTAwODcsIm5iZiI6MTczMDA0OTc4NywicGF0aCI6Ii8yMDgyODAwLzM4MDQ5NTg4Ni0wZTAxYWU1YS0yMGUzLTQxNDYtYTVkNS02NWVmODc5MzE5YjQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjdUMTcyMzA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzkyZjI2YjEyZDE5ZTAwMGUyNzcwZDU1YWY5ZTFkZjcwOTcyNmNlZDI0NmQ4NjkxMjE4ZGIxYjM2MDM3MzQ2OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.wKmrEITTRuhw6fdr_-C4ZaWREcdW8Z6AZtYPxuuzyU8">
> What I was showing by following that,
> 
> Save the file, but do not restart Klipper.
> 
> Run MYTEST again, it didnt update until I manually ran DYNAMIC_MACRO

Sorry for the confusion. That was a typo, and I just fixed it.

> Error loading template 'CHECK_FSENSOR': jinja2.exceptions.TemplateAssertionError: no filter named 'str'

Can you try updating the 3MS using [this](https://3dcoded.github.io/3MS/instructions/#8-updating)?

kaiatkins, 2024-10-29:
[image]
[image]

did update but still fsensor issue

kaiatkins, 2024-10-29:
[image]
:) getting there... but I do keep seeing Unknown command:"SET_TOOL_SYNC"

I pressed the macros just to see what would happen and some came up with errors ( I understand the extruder not up to temperature ones others not so sure)

so adding Klipper.log
[Uploading klippy.log…]()

will start calibration for each of the extruders and get back to you

thanks again for your quick responses and fixes

3DCoded, 2024-10-29:
Your log file didn't fully upload. Can you manually edit `3ms/macros.cfg` and use the find tool to find `|str` in the `CHECK_FSENSOR` macro and replace it with `|string`?

kaiatkins, 2024-10-29:
> Your log file didn't fully upload. Can you manually edit `3ms/macros.cfg` and use the find tool to find `|str` in the `CHECK_FSENSOR` macro and replace it with `|string`?

I got the str error corrected by following what you sent before ( the update and then install.sh) 

I did reupload the log that has the Unknown command:"SET_TOOL_SYNC" incase you want to look

3DCoded, 2024-10-29:
Can you please upload your config folder as a zip?

kaiatkins, 2024-10-30:
[bd-3ms-configs.zip](https://github.com/user-attachments/files/17565878/bd-3ms-configs.zip)

3DCoded, 2024-10-30:
I think I found the issue. The section `[gcode_macro SET_TOOL_SYNC]` is indented by one space. You can either unindent the following lines in `3ms/macros.cfg`:

```
[gcode_macro SET_TOOL_SYNC]
gcode:
```

or update the 3MS configuration.

kaiatkins, 2024-11-01:
[image]
 been working on printing out the hotend pieces and finishing off the hardware things

3DCoded, 2024-11-03:
So far your build looks great!

I'm currently working with the developer of [Happy Hare](https://github.com/moggieuk/Happy-Hare) firmware to migrate the 3MS to Happy Hare. It adds a ton of features and reliability improvements. However, it doesn't natively support this type of setup. The current 3MS software should work fine without a main printer extruder. When I complete migration to Happy Hare in a few months, this type of setup will become unsupported. I would recommend converting your setup to use one extruder as a main extruder, then feeding the other three into it as 3MS steppers. 

I apologize for the inconvenience this may cause. I hope you aren't too far into your build, so this won't be too major of a modification.

kaiatkins, 2024-11-05:
I understand and happy to stay with current if that is ok. If I do see any sales for a direct drive and happen to get one I'll look at switching again. Thanks again for your work and communication during my build.

closed 2024-11-20 by kaiatkins