3DCoded/3MS #5: [FR]: KlipperScreen, endless spool

allenrowand, 2024-08-25:
### What feature do you want to see?

Was wondering if there was any plan to add KlipperScreen support for loading and unloading filament.

My other request would be endless spool, so that (for example) if T0 runs out of filament 3MS would pause, unload T0, load T1, and resume. I'm assuming the best way to implement that would be a runout sensor before each 3MS extruder…

Thanks!

3DCoded, 2024-08-26:
Experimental implementation is complete, and documentation is [here](https://3dcoded.github.io/3MS/experimental/endless/). Let me know if anything needs to be changed.

allenrowand, 2024-08-31:
Would like to talk to you about the project, can you email me at allen@west3d.com?

3DCoded, 2024-09-13:
I've implemented support for Endless Spool for another type of filament sensor setup: a single filament sensor setup, which is what my setup uses. 

After some testing and evaluating the internal logic, a single filament sensor setup should be more reliable than a multi-filament sensor setup (the multi-filament sensor setup requires unloading the empty spool, which could lead to tip shaping issues, while the single filament sensor setup simply rams the new filament into the old without any unload required).

I've updated the documentation with the new recommendations and settings.

3DCoded, 2024-09-27:
Endless Spool is now implemented into the `main` branch.

closed 2024-09-27 by 3DCoded