3DCoded/3MS #7: [FR]: Use of sensorless method to detect filament

yeorinhieut, 2024-09-01:
### What feature do you want to see?

Like ercf, we could use sensorless (tmc2209's feature to detect tolerance) method to check if filament is inserted into extruder. (by checking stop of filament)

Load -> check filament stopness -> extruder&3ms motor synchronization

We can then use 3ms with printers that doesn't have filament sensor on top of extruder! 

+ Can I use skr pico to the 3ms?

3DCoded, 2024-09-03:
Currently, the 3MS doesn't support sensorless filament detection, but I plan on switching the 3MS's firmware to [Happy Hare](https://github.com/moggieuk/Happy-Hare) once support for it is added. Happy Hare firmware does support sensorless filament detection. 

I've added experimental support for the SKR Pico.

3DCoded, 2024-11-11:
Hi @yeorinhieut, I just wanted to let you know that Happy Hare v3 Beta (which supports the 3MS and sensorless filament homing) is now available. See #29.

yeorinhieut, 2024-11-12:
Good news! Thanks for support!

commit 4a375754 (2024-09-03): feat: Add BTT SKR Pico Configuration

Partially resolves #7
closed 2024-11-12 by yeorinhieut