3DCoded/3MS #8: 3MS Development Plan

3DCoded, 2024-09-06:
Here is the development plan for the 3MS project:

- [ ] TPU filament support [LINK](https://3dcoded.github.io/3MS/experimental/dualdrivetpu/)
- [ ] Expanding the no tip shaping feature using skinnydip string reduction
- [ ] 3MS Kits/Pre-assembled units

3DCoded, 2024-11-07:
> "3MS Kits/Pre-assembled systems" - that would be really great! Is this still planned for Q1/2025?

Yes. I'm planning this around February-March 2025.

> The BTT MMU has one disadvantage: limit of 4 colors. As far as I can see, only the A30T board supports more colors? How woild that work once 3MS has moved over to HappyHare?

There are two options in this case:

1. Use multiple BTT MMU boards. If you want to use multiple BTT MMU boards, you can open an issue and I will create a configuration to use multiple to get more colors.
2. I am converting over the existing configs to be compatible with Happy Hare. So far, only a few have been converted, but if you want to check on the progress, the configurations are [here](https://github.com/3DCoded/3MS/tree/main/happy-hare/configurations).

bsafh, 2024-11-07:
Sounds like multiple BTT MMU seem to be the better option, but perhaps I'll better wait for 3MS Kits:-)

ImChrono, 2024-11-08:
@3DCoded FYI, I'm tracking your migration development and adding the missing configurations to my repository. Once everything is ready for testing, it will be ready to merge.

3DCoded, 2024-11-08:
> @3DCoded FYI, I'm tracking your migration development and adding the missing configurations to my repository. Once everything is ready for testing, it will be ready to merge.

@ImChrono I think the only thing for you to migrate would be the Geetech A30 configuration. You can look at the [existing configurations](https://github.com/3DCoded/3MS/tree/main/happy-hare/configurations) to see how to create one. I think the SKR Mini or SKR Pico would be a good starting point.

Filament cutters are already natively supported by HH, so I think that should be an easier migration.

3DCoded, 2024-11-11:
Happy Hare beta v3 is released! See #29.

3DCoded, 2025-05-16:
@bsafh ,

3MS kits are now released! #56 

https://west3d.com/products/3ms-the-simple-and-reliable-multi-material-system-kit-by-3dcoded