3DCoded/3MS #19: [Bug]: Print Pauses after Prugeline

Mawe066, 2024-10-14:
### What happened?

Hey, the print pauses automaticly after the purge line. Maybe I am missing something in the slicer configuration but after 3MS starts and reads the T value, the print pauses and i have to manually resume it. Dont know if it is the same after another toolchange, havent done my testing so far.

### Klippy Log

[klippy.log](https://github.com/user-attachments/files/17363844/klippy.log)

3DCoded, 2024-10-14:
Does a prompt show up in Mainsail/Fluidd/KlipperScreen when this happens?

Mawe066, 2024-10-14:
[image]

The "Print is not paused, resume aborted" comes from an work around i have tried without succsess. Seems like it makes the purgeline, then checks again the filament sensor and pauses the print. The Start command is placed before the purge line, like the documentation said. I am using orca slicer. Except for the two thing to add in the maschine gcode, there is nothing more that needs modifications?

3DCoded, 2024-10-14:
Did the print just skip from starting to ending without actually printing anything?

Mawe066, 2024-10-14:
No, I printed flow calibration in a single color.

Right at the beginning or after the purge line? Didn't see the need of an clear pause before it actually paused

3DCoded, 2024-10-14:
Generally, adding a `CLEAR_PAUSE` before even your heating/homing commands is a good practice for a Klipper start G-code (with or without the 3MS). The `CLEAR_PAUSE` command resets Klipper's internal pause state. 

When the 3MS loads filament, do you hear a clicking noise? Does the filament actually get loaded to your filament sensor?

Mawe066, 2024-10-14:
The filament is loaded. Like I said, it makes the purge line and then right into the pause state

3DCoded, 2024-10-14:
This sounds like your 3MS's `fsensor_delay` setting needs to be increased. Can you try increasing your `fsensor_delay` to something large, like 10,000ms?
```
SET_MMMS_SETTINGS FSENSOR_DELAY=10000
```

If that works, slowly decrease your `fsensor_delay` to be as low as possible while still not pausing the print automatically.

To save permanently, edit your `3ms/settings.cfg`:
```
[gcode_macro MMMS_SETTINGS]
other settings...
fsensor_delay: 10000
```

Mawe066, 2024-10-14:
I set the delay to 10.000 and it still pauses after the purgeline. I eben tried to increse the settings in my filament sensor itself. Still no changes. i have an BTT SFS 2.0.

Mawe066, 2024-10-15:
Another thing, not bug related.  Is it possible to pre load the initial extruder before loading via 3ms? My specific setup reqiuries this :D

Mawe066, 2024-10-15:
Hey, i have another question i cant find anything about in the documentation. In orca slicer i have set an cooling tube position to 35mm and filament park position to 160 what is in between my sensor and my direct drive extruder. Is 3ms ment to load to the parking psoition and let orca/super slicer handle the rest? If yes, i need another 160 millimeters to load with my initial tool. If i ignore my parking position, and load directly into the nozzle, i have an massive blob. For now i cant figure out how the set the values right in my slicer and my loading and unloading distances

ImChrono, 2024-10-15:
Just for reference, I also use Orca Slicer. I've set the cooling tube position and length to 0, allowing the 3ms macros to handle the task. My setup is slightly different, as I've positioned the filament sensor after the extruder gears. However, I don’t think this makes a significant difference, as long as the values are correctly configured in the macro settings.

Mawe066, 2024-10-15:
I have 4 extruders that lead through the sensor to the extruder that's directly attached to the hot end. But how does the tip shaping works without the slicer settings?

ImChrono, 2024-10-15:
I'm sorry, but I can't help with that aspect. I've completely bypassed filament tip shaping. Initially, I only printed with PLA, and now I've added a filament cutter, which I’ve been using reliably ever since.

3DCoded, 2024-10-15:
> I have 4 extruders that lead through the sensor to the extruder that's directly attached to the hot end. But how does the tip shaping works without the slicer settings?

You need slicer settings for tip shaping. The [Materials table](https://3dcoded.github.io/3MS/guides/materials/) can show you which filaments need tip shaping, some recommended tip shaping settings, and if they support [No tip shaping](https://3dcoded.github.io/3MS/guides/notip/). I am also currently working on a faster tip shaping method.

3DCoded, 2024-10-15:
> Hey, i have another question i cant find anything about in the documentation. In orca slicer i have set an cooling tube position to 35mm and filament park position to 160 what is in between my [...]

Yes. The 3MS is meant to load the the parking position and let the slicer handle the rest. The slicer should automatically handle the extra loading. Usually, you only need to set the 3MS's `load_distance` to 5-15mm more than your `unload_distance`. To fix the blob, you can set your slicer's `Printer settings` -> `Multimaterial` -> `Extra loading distance` to a negative number, like `-20`.

Mawe066, 2024-10-15:
But when I load it to the parking position in the initial loading, then there is 160mm missing in my case. Is there a possibility to implement a macro or something else like that into the initial loading process?

3DCoded, 2024-10-15:
The proper sequence of events for a tool change is:
- Slicer performs tip shaping
- Slicer unloads to parking position
- 3MS unloads from parking position
- 3MS loads to parking position
- Slicer loads from parking position
- Slicer performs purging

Does your tool change follow this sequence? 

Also, I have a few more questions:
- What did you set the `Extra loading distance` setting to?
- What slicer are you using? 
- What are your 3MS `unload_distance` and `load_distance` settings?

Also, for the earlier problem with the filament motion sensor, you can add these lines before and after your `MMMS_START` call in your Start G-Code:
```
SET_FILAMENT_SENSOR SENSOR=encoder_sensor ENABLE=0
MMMS_START
SET_FILAMENT_SENSOR SENSOR=encoder_sensor ENABLE=1
```
This will disable your filament motion sensor during `MMMS_START`, then re-enable it after.

Mawe066, 2024-10-15:
My Question is specifically for the first load, not a color change. How does it handle the, in my case, missing 160mm filament from the parking position to the tip of the nozzle, since there is no tip shaping and unloading required. I understand how the color change works, but that's not my question.

Mawe066, 2024-10-16:
I think you can close this thread now. I have found an solution for my specific build related problem. I just added the missing 160mm of filament after the loading sequence in the start code. Works like a charm now.

closed 2024-10-16 by 3DCoded