3DCoded/3MS #20: Manual filament cutter

ImChrono, 2024-10-15:
### What feature do you want to see?

Hi there,
I'm opening this feature request to track and provide suggestions for a manual filament cutter feature. I've already made some local modifications to the project, and it's working great.

What I've done is create a new extruder for bed slinger-type 3D printers, incorporating a blade and a specific filament cutter position. Before I commit my work, I wanted to explore ways to further improve the workflow and, more importantly, assess whether such a feature is truly needed.

I've set up a parking position near the filament cutter, and during each filament change, I move the toolhead to this position and activate a custom macro. This macro executes four quick movements to trigger the filament cutter, allowing the filament to be changed as usual.

What I'd like to add is a custom retraction to save filament and, if possible, make the filament "poop" out instead of relying on the purge tower. I'll also share some photos of the setup and process for reference.

[image]

https://github.com/user-attachments/assets/3de968df-6425-4234-8b5b-bedcf8b73a23

3DCoded, 2024-10-15:
I've experimented with filament cutters in the past, and have had a hard time finding a good design for my printer (Prusa MK3S+), so I haven't been able to do much testing.

From my limited testing with filament cutters, I've come up with some pro's and con's.
**Pros:**
- Faster
- More reliable
- No need for tip tuning

**Cons:**
- Need printer-specific hardware
- Extra waste is generated

I'm also working on a faster/simpler tip shaping algorithm that should also make tip shaping much easier to tune. 

With that said, if you have a working setup, I'm open to pull requests. If you do a PR, to keep the 3MS software modular, please put the filament cutter configuration in either its own config file (e.g. `cutter.cfg`) or its own folder with multiple files inside (e.g. `cutter/settings.cfg` and `cutter/macros.cfg`).

I am planning on migrating to [Happy Hare](https://github.com/moggieuk/Happy-Hare) firmware as soon as support for 3MS-style systems is added, which does support filament cutters (and much more). I'd recommend looking at its source code for the extra retraction before cutting, and custom purge locations.

ImChrono, 2024-10-16:
Thank you for this awesome project!
As you mentioned, filament cutting is more reliable and faster. While I’m not a fan of waste either, for now, it’s the only dependable solution that works with stringy filaments and multi-material setups. That said, I'm working on rebasing the macros and making them modular, so I can commit the changes to the main branch. I still need to figure out how to make the MMMS_UNLOAD macro read and use the new module only if it’s actually enabled—once that’s done, I'll commit a first version.

The next step would be to add filament retraction.

Off-topic: Happy Hare is a very cool project! But the last time I checked, the 3MS-style configuration was still theoretical. In my opinion, using multiple motors with multiple extruders is more cost-effective than building an MMU. Older printers like the Ender 3 are cheaper nowadays and can serve as a great starting point for making a 3MS-style setup with minimal additional accessories.

3DCoded, 2024-10-16:
To check if a module exists, you can check if one of its macros exists. For example, if one of your macros is named `CUT_FILAMENT`, you could use:

```
{% if "gcode_macro CUT_FILAMENT" in printer %}
; Filament cutting is enabled
{% else %}
; Filament cutting is disabled
{% endif %}
```

I asked the developer of Happy Hare about 3MS-style systems ([#371](https://github.com/moggieuk/Happy-Hare/issues/371)), and they said that the functionality was implemented, but the configuration isn't exposed. I looked through the Happy Hare code, and I'm pretty sure the `development` branch has that code. I'm currently waiting for it to be more tested and documented before moving the 3MS over to Happy Hare.

3DCoded, 2024-10-16:
For filament retraction, you can probably edit the `Toolchange` macro and add a retract command at the start, using the same method I recommended earlier:
```
{% if "gcode_macro "RETRACT_BEFORE_CUT" in printer %}
; Filament cutting is enabled
RETRACT_BEFORE_CUT
{% endif %}
```

Because there are many different filament cutter designs available, I would recommend making the settings configurable, like the `MMMS_SETTINGS` macros in `3ms/settings.cfg`.

ImChrono, 2024-10-16:
I've just committed the first revision to my own repository. This configuration includes a decompress pin on the X-axis. Before merging and submitting a PR, I’ll complete the "manual mode" filament cutter so it can be configured for both the X and Y axes, making it compatible with CoreXY systems as well. I’m aware that servo mode is an option, and I plan to work on that too. I still need to update a few files, and once I'm done, I'll update you.

3DCoded, 2024-10-16:
Your implementation looks good! I'd recommend a couple things in its current state:
- Implementing `SET_CUTTER_SETTINGS` and `GET_CUTTER_SETTINGS` macros for easier settings changes
- Making the number of cut repetitions configurable

ImChrono, 2024-10-16:
I've added setters and getters and parameterized the cut retries. I'm currently testing it with a long print, and I’ll check the results tomorrow.

3DCoded, 2024-10-16:
The setters/getters look good. Thank you for using the same code conventions as the rest of the 3MS code.

On a side note, could you please post the link to that Y-splitter?

ImChrono, 2024-10-17:
You’ll need 5 PC4-M10 connectors. Be careful not to push the Bowden tubes too far into the four inputs, as this can block the filament path. The model itself already includes a support structure. I recommend printing the main model in ABS or PETG, and the support pin in a different material to make removal easier.

Please note, I’ll be publishing a newer version of the model soon that addresses some issues I encountered during printing.
[4-Way Y-Splitter Model](https://www.printables.com/model/1042279-4-way-y-splitter)

3DCoded, 2024-10-17:
Thank you! I've been trying to find a good 4-way splitter. I'll print that out soon. How did the test print go?

ImChrono, 2024-10-17:
Still printing, looks good for now. 
[image]

ImChrono, 2024-10-18:
Unfortunately, the print failed due to issues with the Automatic Pause. I wasn't home, and the printer timed out.

The good news is that the cutting macros are working great!

Off-topic:
I’m still investigating the cause of the pause issue. I’m not sure if it’s related to timing (I’ve tested up to 10 seconds). I’ve also noticed that my Raspberry Pi occasionally detects undervoltage, so it could be a system-related issue.
However, I bypassed the problem by editing the settings as follows:

[filament_switch_sensor hsensor]
pause_on_runout: False
switch_pin: EBBCan:PB6

Keep in mind that I have two endstops. The one I'm using for the 3MS macros is positioned after the extruder gears and is not being used as a filament runout sensor, so automatic pause isn’t a feature I need.

3DCoded, 2024-10-18:
The automatic pause should only happen if the filament sensor shows that the color swap failed. If this keeps happening, let me know with a klippy.log, and I can diagnose the problem.

ImChrono, 2024-10-18:
I've attached the klippy.log. I hope you can find the culprit.
[klippy.zip](https://github.com/user-attachments/files/17440473/klippy.zip)

3DCoded, 2024-10-18:
If you can catch this happening, I have a few main ideas:

1. If the `klippy.log` indicates that your `fsensor` (not `hsensor`) was triggered, disable the `fsensor` during a print and test again
2. If the `klippy.log` indicates that your `hsensor` was triggered, increase your 3MS `fsensor_delay` setting and test again
3. If the color swap actually failed (e.g. the previous filament failed to unload or the next filament failed to load), the automatic pause was expected behavior. Either your LCD or Mainsail/Fluidd/KS should display a message in this case.

This `klippy.log` indicates point two, so I'd recommend increasing your `fsensor_delay` setting. If this happens again, look back over the three points and follow the suggested solution.

ImChrono, 2024-10-18:
What I don't understand is that the majority of filament swaps work perfectly, but occasionally the hsensor got triggered (or at least, I assume that's what's happened). The fsensor is fine—I’ve tried printing with the sensor disabled, and the issue still occurred.

Another thing to note: I’m currently printing with the setting hsensor: pause_on_runout: False and fsensor disabled, and no issues have occurred so far. However, it could just be a placebo effect, as I restarted the firmware before this test.

ImChrono, 2024-10-19:
This time, the print was successful!

[image]

ImChrono, 2024-10-22:
Another successful print! This is the first print with filament retraction enabled. I still need to fix a few issues, and then I'll commit the first revision.

[image]

3DCoded, 2024-10-22:
Those prints look great! I'm looking forward to your PR.

3DCoded, 2024-10-31:
Closing as the PR is merged.

closed 2024-10-31 by 3DCoded