3DCoded/3MS #24: Filament feeder splitter

ImChrono, 2024-10-20:
I'm opening this thread to track and discuss filament splitter for the toolhead. Below is the link to my latest revision:

https://www.printables.com/model/1042279-no-catch-4-way-y-splitter-pc4-m10

This updated revision is designed for easier installation and printing. No supports are needed—just follow the slicer configuration provided.

[image]

Improvements and alternatives are welcome

3DCoded, 2024-10-20:
Thank you for designing this. I've been using the original version on my setup for a few days now and have had no problems so far. Utilizing the `Bridge counterbore holes` setting is a good way to remove supports on prints.

ImChrono, 2024-10-20:
Good to hear it's working well for you too. This new revision is designed to be more forgiving when installing the Bowden tubes, ensuring they’re aligned at the same distance. It also helps reduce unwanted pressure that can occur if the feeder motor isn't properly aligned with the extruder motor.