3DCoded/3MS #28: [Bug]: Klipper pressure_,advance stops working after 3MS install

deemax-real, 2024-11-09:
### What happened?

Hi, after 3MS install Klipper pressure advance have mo impact on the print. Pressure advance and smooth time windows disappeared from mainsail gui and even gcode pressure advance have no impact on print (round corners). Also extruder acceleration from printer.cfg stops working (now extruder works with same ACC as axes). Can provide logs before and after 3MS installation if needed.

### Klippy Log

Z

deemax-real, 2024-11-09:
[config-20241109-174017.zip](https://github.com/user-attachments/files/17688828/config-20241109-174017.zip)
[klippy.log](https://github.com/user-attachments/files/17688829/klippy.log)

3ms controller steppers.cfg are for Ender 4.2.2 motherobard under btt_skr_mini_e3_v2

3DCoded, 2024-11-09:
I can't see anything wrong with your config. Can you try a PA calibration print?

deemax-real, 2024-11-09:
Printer seems to ignore pressure advance setting and even command from console during printing. Can send photo of model printed with pa working and some model from today wit obvious overextruded corners. Even silly high pa value not change anything :)

3DCoded, 2024-11-09:
Does the issue still occur if you comment out this line in your `printer.cfg`:

```
[include 3ms/main.cfg]
```

and add a test Dynamic Macro as explained [here](https://3dcoded.github.io/DynamicMacros/setup/#configuration)?

deemax-real, 2024-11-09:
PA and Smooth time coming back after commenting 3ms/main.cfg

but....
Dynamic Macros test done nothing. No response for MYTEST :/ 
[image]
[image]

deemax-real, 2024-11-09:
After some tests I've find dynamic macros test didn't work with m117 command :/
After change in dynamic.cfg I'm getting response in console.
[image]
[image]

3DCoded, 2024-11-09:
This is really confusing. Can you try the following steps and see if PA works?

1. Uncomment the `[include 3ms/main.cfg]` line in your `printer.cfg`
2. In `3ms/main.cfg`, comment out the `[dynamicmacros 3ms]` section.

Also, I would recommend replacing your main `[mcu]` configuration with a `/dev/serial/by-id` path (it's currently `/dev/ttyAMA0`), so it doesn't get confused with the 3MS board.

deemax-real, 2024-11-09:
After uncommenting 3ms/main.cfg PA gone from gui 

Pi connected by serial with printer mb so can't change to usb connection. Pi seen both mcu's without any problem and everything works (3ms motor spins and sync with main extruder motor) so don't think connection is a problem but later today or tomorrow I can try back to us connection for both mcu's.

3DCoded, 2024-11-09:
> 2. In 3ms/main.cfg, comment out the [dynamicmacros 3ms] section.

Did you try this?

deemax-real, 2024-11-09:
Yes. No change

3DCoded, 2024-11-09:
Ok. That helps to narrow down the issue. I'm guessing it has to do with how the `steppers.cfg` file is set up (probably my fault). I'll have to check this.

deemax-real, 2024-11-09:
Happy to try any solution :)

3DCoded, 2024-11-09:
Hmm... That was the first configuration I used with the 3MS. I think it has to do with the serial port. I would recommend converting your serial to use `/dev/serial/by-id` for both the 3MS and your main MCU.

deemax-real, 2024-11-09:
Trying do this now but can't get 2 usb IDs from terminal. Don't know why.

deemax-real, 2024-11-09:
Usb sorted. 
Both mcu's connected by usb to pi
All changes/comments reverted.
No PA and Smooth time :(

[image]
[image]
[image]

3DCoded, 2024-11-09:
I think we're going to have to go through `3ms/main.cfg` and figure out what is causing the error. So far, it seems like the `steppers.cfg` is the issue. What happens if you comment out the `[include ./controllers/.../steppers.cfg]` line in `3ms/main.cfg`?

Thank you for your patience as we work through this error!

deemax-real, 2024-11-09:
After commenting steppers.cfg PA comes back. 
[image]

I'm happy to help (and desperate to find a solution) as I want my ender printing in colours for my sons :)

3DCoded, 2024-11-09:
That helps to narrow down the issue. I have an idea:

In the `3ms/controllers/skr_mini_e3_v2/steppers.cfg`, can you comment out all the `extruder_stepper` sections?

If that doesn't work, can you try also commenting out the `[mcu 3ms]` section?

Let me know if either of those work (the 3MS won't work in this case but it should at least give another pointer to find the root cause).

deemax-real, 2024-11-09:
[image]
[image]

3DCoded, 2024-11-11:
Aside from updating Klipper, I don't really have other ideas. `extruder_stepper`s are part of standard Klipper, so I expected them to "just work" with any Klipper setup. If you can remove the 3MS configuration and add an `extruder_stepper` configuration, and PA still disappears, you might be able to get some help from the [Klipper Discourse](https://klipper.discourse.group/).

Alternatively, Happy Hare v3 beta just got released, which now supports the 3MS. It uses an entirely different configuration system, so if you want to become a beta tester, see #29.

deemax-real, 2024-11-11:
Hi, great news about Happy Hare beta. I'll lean into this way and we'll see if  PA working. 
Happy to be a beta tester.

3DCoded, 2024-11-15:
Closing this issue as it's been moved to the HH Discord.

closed 2024-11-15 by 3DCoded