3DCoded/3MS #30: [MCU]: Fly-D7

kintarojs, 2024-11-15:
### What kind of controller is this?

extra MCU (MCU used specifically for the 3MS)

### Controller name

Fly-D7

### More information

https://mellow.klipper.cn/en/docs/category/fly-d7

3DCoded, 2024-11-15:
Hi @kintarojs,

Thank you for your controller request. I have added support for the Mellow Fly D7 board in the existing 3MS firmware and in Happy Hare. Documentation will be available soon.

3DCoded, 2024-11-16:
Sorry for the errors. Thank you for your fixed version. I'll update this.

3DCoded, 2024-11-16:
I've fixed as suggested, and the documentation is now available [here](https://3dcoded.github.io/3MS/setup/controllers/mellowflyd7/). Let me know if there's anything I need to change.

closed 2024-11-29 by 3DCoded