3DCoded/3MS #32: [DOCS]: Supplement the documentation on BTT MMB

Jager-f, 2024-12-04:
### What kind of documentation issue is this?

- [ ] Typo
- [ ] Incorrect information
- [ ] Unclear information
- [X] Missing information
- [ ] Contradictory information

### How do you think the documentation should be improved?

In the current version of the document, when power is supplied to the HVIN, the board will not be powered and will not start. It is mandatory to install the jumper HUSB, to power the processor from USB.

3DCoded, 2024-12-04:
Hi @Jager-f ,

Thank you for finding this issue in the documentation. I have updated the beta documentation [here](https://3dcoded.github.io/3MS-docs/setup/controllers/bttmmb/). When Happy Hare v3 gets fully released, this will replace the current documentation.

closed 2024-12-04 by 3DCoded

Jager-f, 2024-12-08:
These two lines need to be swapped. The LED will light up only after connecting the USB

If the MMB lights up, you wired it correctly!

Finally, plug the MMB into your Klipper host with the cable that came with it.

3DCoded, 2024-12-09:
Hi @Jager-f ,

I have switched the last two lines in the **main** documentation (the beta docs are now fully migrated to the main docs).

Jager-f, 2024-12-09:
I made a compact and lightweight Y-splitter, with tubes fastened with 1 screw for 3MS) https://www.printables.com/model/1092036-4-tube-y-splitter-for-mmu

3DCoded, 2024-12-09:
> I made a compact and lightweight Y-splitter, with tubes fastened with 1 screw for 3MS) https://www.printables.com/model/1092036-4-tube-y-splitter-for-mmu

That's an interesting design. This is good timing too, as I'm trying to figure out the easiest to print and most reliable Y-spliters. I may look into this as I already have the required hardware.

3DCoded, 2024-12-09:
Also, @Jager-f ,

If you want more people to test out your Y-splitter, you can post the link on the 3MS discord and the 3MS channel at the HH discord. Both links are in the homepage of the docs.