3DCoded/3MS #35: [MCU]: BTT Kraken V1.0

t7git, 2024-12-08:
### What kind of controller is this?

main MCU (using spare stepper ports on existing MCU)

### Controller name

BTT Kraken V1.0

### More information

This is gutting out an old Neptune 4 Max only using the hotend/extruder and obviously the bones from the Neptune 4 Max) I use my old laptop it's an Intel i5 processor for my Klipper host. Then I use a BTT Kraken as my main board. (60V on the stepper motors, 24V on the Bed Power, 24V on the MCU power) The stepper motors are X, Y, (corrupted not used), 3MS0, Z1, Z, 3MS1, E

Then on a secondary MCU it's the BTT Manta M8P and I have two 2209 stepper motor drivers installed (I own five 2209 but haven't purchased more extruders yet (I only have 4 right now)). The two separate 2209 on the Manta M8P are 3MS3 and 3MS2. 

The Manta's type C connection doesn't work properly so I use the UART to connect the Manta to the Klipper host. I also use a CAN to USB adapter and own the EBB42 that is coming in the mail today that is replacing a broken one.. (BTT's quality control is crazy.. never returned so many things in my life) The Manta using UART is using a Serial to USB adapter. The new EBB42 is going to replace the extruder, hotend thermistor, filament run out sensor, inductive probe, servo, and hotend heater.

3DCoded, 2024-12-09:
Hi @t7git ,

Just to confirm the steppers:

- `3ms0` is `S4`
- `3ms1` is `S7`
- `3ms2` is Manta `X`
- `3ms3` is Manta `Y`

I'm going to merge #36 into this to make it easier to keep track of these.

3DCoded, 2024-12-10:
Hi @t7git ,

I have added the configuration. It is available [here](https://github.com/3DCoded/3MS/tree/custom-configs/configs/35) .