3DCoded/3MS #37: Migration to Happy Hare v3

3DCoded, 2024-12-09:
With the final release of Happy Hare v3.0, the 3MS will now entirely migrate to Happy Hare firmware. The old docs and software will be saved in the `legacy` and `docs-legacy` branches, but it his **highly** recommended to switch to Happy Hare firmware. If you already used the Happy Hare beta, all you need to do is:

```
cd ~/Happy-Hare
./install.sh -b main
```

If this is a new installation, follow [this guide](https://github.com/moggieuk/Happy-Hare/wiki/Quick-Start-3MS) .

Also, please join the 3MS channel in the Happy Hare Discord:

https://discord.gg/8nTTEZeuvb

and the 3MS Discord:

[image]

brentdeacon23, 2024-12-10:
I am able to clone the HH repo but I cant run install.sh, im assuming this is due to the K1's "klipper" version. Any insight?

3DCoded, 2024-12-10:
> I am able to clone the HH repo but I cant run install.sh, im assuming this is due to the K1's "klipper" version. Any insight?

Hi @brentdeacon23 ,

Someone on the #3ms channel in the HH Discord was able to setup HH on a K1. From what I can tell, the two main things that they did that fixed it was:

1. Install bash
2. Install python3 with:

    ```
    opkg install python3
    ```

closed 2024-12-18 by 3DCoded