3DCoded/3MS #40: [MCU]: MKS Monster8

Mawe066, 2024-12-12:
### What kind of controller is this?

main MCU (using spare stepper ports on existing MCU)

### Controller name

MKS Monster8

### More information

For the usage in Happy Hare with no exra MCU. All Steppers are inside the existing MCU

3DCoded, 2024-12-12:
Hi @Mawe066 ,

Do you want the last 4 ports to be used for 3MS extruders (E1, E2, E3, and E4)?

Mawe066, 2024-12-12:
Yes, I have an direct drive and 4 feeding extruders

3DCoded, 2024-12-12:
Hello @Mawe066 ,

I have created the configs using the new configuration generator tool (docs coming soon). Here are the configuration files:

[mmu.zip](https://github.com/user-attachments/files/18117653/mmu.zip)

Mawe066, 2024-12-13:
that was quick, thank you!

closed 2024-12-13 by 3DCoded

3DCoded, 2024-12-13:
Hi @Mawe066 ,

I have found an error in the `mmu_hardware.cfg` I generated. Here's the updated one:

[mmu_hardware.cfg.zip](https://github.com/user-attachments/files/18126428/mmu_hardware.cfg.zip)

Mawe066, 2024-12-13:
What was that error? Maybe I already fixed it. I needed a few more parameters to get no errors.

Mawe066, 2024-12-13:
[image]

Got errors in missing parameters on interpolate and sense resistor. Had to add these

Mawe066, 2024-12-13:
Ah and I forgot to mention, I use 2209 steppers. In your config were 2208. Maybe add a question in the request template for this

3DCoded, 2024-12-13:
Good idea. I've updated the form so future requests will include the driver question.